MannLabs/alphatims #117: Loading data that were not collected using ddaPASEF or diaPASEF

mastopoly, 2021-05-03:
**Is your feature request related to a problem? Please describe.**
Data acquired using a TIMS method, but not a PASEF method enabled cannot be loaded.  The GUI gives the error :"Scan mode is not ddaPASEF or diaPASEF" and said the file was corrupted. I am running on windows.

I also tried on command line by running the following:
`alphatims.bruker.TimsTOF(r'C:\Users\Mast\Desktop\timstof_python\analysisname.d')`

The following error occurred: 
Traceback (most recent call last):

  File "<ipython-input-29-58b1171ccd4d>", line 1, in <module>
    atb.TimsTOF(r'C:\Users\Mast\Desktop\timstof_python\analysisname.d')

  File "C:\Users\Mast\AppData\Roaming\Python\Python38\site-packages\alphatims\bruker.py", line 777, in __init__
    self._import_data_from_d_folder(

  File "C:\Users\Mast\AppData\Roaming\Python\Python38\site-packages\alphatims\bruker.py", line 817, in _import_data_from_d_folder
    ) = read_bruker_sql(bruker_d_folder_name)

  File "C:\Users\Mast\AppData\Roaming\Python\Python38\site-packages\alphatims\bruker.py", line 185, in read_bruker_sql
    with sqlite3.connect(

OperationalError: unable to open database file

**Describe the solution you would like**
I would like to be able to load a file that has TIMS data, but not MS/MS data. if this is possible.

swillems, 2021-05-04:
Dear mastapoly,

As our lab mostly focusses on MS/MS data, I unfortunately have no experience with pure TIMS data. Looking at your error messages I think that the data format in which Bruker stores the TIMS data is different than the way they store LC-TIMS-MSMS data as produced by their timsTOF Pro instrument. If this is indeed the case, I am afraid that AlphaTims is not the right tool for you. However, if your .d folder contains both the "analysis.tdf" and "analysis.tdf_bin" files, I could briefly see if this format is compatabile with AlphaTims.

Best,
Sander

swillems, 2021-05-04:
Dear mastapoly,

Thank you for providing the raw data in a private message. For future corresponding (which does not involve unpublished data) I suggest we communicate via this issue so other people will similar issues can be helped as well.
It looks like the data you provided is actually compatible for the largest part. I quickly tested if I can update AlphaTims based on the difference in format and this actually seems to work quite nicely as well. I will update the source code where needed and I will aim to release version 0.2.6 somewhere this week. Please note that this use case falls somewhat outside the scope of AlphaTims and we will hence not guarantee it to run completely bug-free at this point in time.

Best,
Sander

mastopoly, 2021-05-04:
Sander, 
Thank you for taking a look at this. I will install the updated packages and let you if things work. In addition, I would be happy to have provided the file on github and the data I sent is actually already published, I couldn't upload it to github on this forum because it was too large. If you would like to include the file in an example or to support open source. But if you would like to upload it for the sake of others, that is okay.

swillems, 2021-05-05:
Dear mastapoly,

Now worries about the dataset, just wanted to make sure that people who read this issue in the future can follow along and know what we are talking about. If you want you can provide a link to the paper as reference.
 
More important, I just released version 0.2.6, which should solve your issue.  Please check it out and let me kow if this works as expected so we can close this issue.

Extra info:
* This issue was caused because AlphaTims raises an error when the `fragments` table in `analysis.df` does not contain an 8 or 9 in the `MsMsType` column, indicating ddaPASEF or diaPASEF scans. Without this, there is no `diaFragmentFrameMsMsInfo` of `PasefFrameMsMsInfo` table, meaning that no `quad_indptr` array and other associated arrrays can be defined.
* Solved by creating a dummy `MsMsInfo` table stating that in frame 0 from start scan 0 to end scan 0 there is a precursor with index 0 and quad selection of 0 m/z present. This results in empty sliders in the GUI, so the GUI has been updated to disable usage of quad information with default 0-1 sliders.
* This is released for all the GUIs on all Operating Systems and on PyPi as well.
* NOTE: This functionality ifalls outside of AlphaTims main scopre and will not be a primary focus for future development.

closed 2021-05-19 by swillems