MannLabs/alphatims #127: Error during file import

flmeier, 2021-05-28:
**Describe the bug**
I get the Type Error pasted below when loading an older diaPASEF file. 

**To Reproduce**

```
import alphatims.utils
import alphatims.bruker
import alphatims.plotting
import importlib

bruker_d_folder_name = "20190319_tims01_FlMe_SA_BSA_45min_diaPASEF_C1_01_5714.d"
data = alphatims.bruker.TimsTOF(bruker_d_folder_name)
```

Temporary download link: [datashare](https://datashare.biochem.mpg.de/s/hqUGfcHv4QOgE1Z)

**Expected behavior**
Successful file import. 

**Logs**

> 2021-05-28 20:50:55> Importing data from 20190319_tims01_FlMe_SA_BSA_45min_diaPASEF_C1_01_5714.d
> 2021-05-28 20:50:55> Reading frame metadata for 20190319_tims01_FlMe_SA_BSA_45min_diaPASEF_C1_01_5714.d
> 2021-05-28 20:50:57> Reading 25,301 frames with 241,304,631 detector strikes for 20190319_tims01_FlMe_SA_BSA_45min_diaPASEF_C1_01_5714.d
> 100%|███████████████████████████████████████████████████████████████████████████| 25301/25301 [01:41<00:00, 248.12it/s]
> 2021-05-28 20:52:39> Indexing 20190319_tims01_FlMe_SA_BSA_45min_diaPASEF_C1_01_5714.d...
> 2021-05-28 20:52:39> Opening handle for 20190319_tims01_FlMe_SA_BSA_45min_diaPASEF_C1_01_5714.d
> 2021-05-28 20:52:39> Fetching mobility values from 20190319_tims01_FlMe_SA_BSA_45min_diaPASEF_C1_01_5714.d
> 2021-05-28 20:52:39> Closing handle for 20190319_tims01_FlMe_SA_BSA_45min_diaPASEF_C1_01_5714.d
> 2021-05-28 20:52:39> Opening handle for 20190319_tims01_FlMe_SA_BSA_45min_diaPASEF_C1_01_5714.d
> 2021-05-28 20:52:39> Fetching mz values from 20190319_tims01_FlMe_SA_BSA_45min_diaPASEF_C1_01_5714.d
> 2021-05-28 20:52:40> Closing handle for 20190319_tims01_FlMe_SA_BSA_45min_diaPASEF_C1_01_5714.d

```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-7-98bb86a635c5> in <module>
      1 # reload()
      2 bruker_d_folder_name = "20190319_tims01_FlMe_SA_BSA_45min_diaPASEF_C1_01_5714.d"
----> 3 data = alphatims.bruker.TimsTOF(bruker_d_folder_name)

d:\jupyter\alphatims\alphatims\alphatims\bruker.py in __init__(self, bruker_d_folder_name, mz_estimation_from_frame, mobility_estimation_from_frame, slice_as_dataframe)
    893         logging.info(f"Importing data from {bruker_d_folder_name}")
    894         if bruker_d_folder_name.endswith(".d"):
--> 895             self._import_data_from_d_folder(
    896                 bruker_d_folder_name,
    897                 mz_estimation_from_frame,

d:\jupyter\alphatims\alphatims\alphatims\bruker.py in _import_data_from_d_folder(self, bruker_d_folder_name, mz_estimation_from_frame, mobility_estimation_from_frame)
   1015                 tof_intercept + tof_slope * np.arange(self.tof_max_index)
   1016             )**2
-> 1017         self._parse_quad_indptr()
   1018         self._intensity_min_value = int(np.min(self.intensity_values))
   1019         self._intensity_max_value = int(np.max(self.intensity_values))

d:\jupyter\alphatims\alphatims\alphatims\bruker.py in _parse_quad_indptr(self)
   1667             ]
   1668         )
-> 1669         self._precursor_max_index = int(np.max(self.precursor_indices)) + 1
   1670 
   1671     def index_precursors(

<__array_function__ internals> in amax(*args, **kwargs)

~\AppData\Local\conda\conda\envs\alphatims\lib\site-packages\numpy\core\fromnumeric.py in amax(a, axis, out, keepdims, initial, where)
   2731     5
   2732     """
-> 2733     return _wrapreduction(a, np.maximum, 'max', axis, None, out,
   2734                           keepdims=keepdims, initial=initial, where=where)
   2735 

~\AppData\Local\conda\conda\envs\alphatims\lib\site-packages\numpy\core\fromnumeric.py in _wrapreduction(obj, ufunc, method, axis, dtype, out, **kwargs)
     85                 return reduction(axis=axis, out=out, **passkwargs)
     86 
---> 87     return ufunc.reduce(obj, axis, dtype, out, **passkwargs)
     88 
     89 

TypeError: '>=' not supported between instances of 'int' and 'NoneType'
```

**Version (please complete the following information):**
 - Installation Type: Developer

2021-05-28 20:45:27> Platform information:
2021-05-28 20:45:27> system        - Windows
2021-05-28 20:45:27> release       - 10
2021-05-28 20:45:27> version       - 10.0.19041
2021-05-28 20:45:27> machine       - AMD64
2021-05-28 20:45:27> processor     - Intel64 Family 6 Model 158 Stepping 9, GenuineIntel
2021-05-28 20:45:27> cpu count     - 8
2021-05-28 20:45:27> cpu frequency - 2901.00 Mhz
2021-05-28 20:45:27> ram           - 25.9/31.8 Gb (available/total)

2021-05-28 20:45:27> Python information:
2021-05-28 20:45:27> alphatims  - 0.2.7
2021-05-28 20:45:27> bokeh      - 2.2.3
2021-05-28 20:45:27> click      - 7.1.2
2021-05-28 20:45:27> datashader - 0.12.1
2021-05-28 20:45:27> h5py       - 3.2.1
2021-05-28 20:45:27> hvplot     - 0.7.1
2021-05-28 20:45:27> numba      - 0.53.0
2021-05-28 20:45:27> pandas     - 1.2.3
2021-05-28 20:45:27> psutil     - 5.8.0
2021-05-28 20:45:27> python     - 3.8.8
2021-05-28 20:45:27> python-lzf - 0.2.4
2021-05-28 20:45:27> pyzstd     - 0.14.3
2021-05-28 20:45:27> selenium   - 3.141.0
2021-05-28 20:45:27> tqdm       - 4.59.0

swillems, 2021-05-29:
Dear Florian,

This is an interesting bug which is probably quite unique to your sample. Luckily, it has a relatively easy fix if you make a few assumptions which seems acceptable given the fact that few people will ever encounter similar samples.

In short, here is my interpretation of what happened: This is likely one of the very first diaPASEF acquisitions, which was probably made possible with Bruker's otofControl. However, even though the otofControl allowed the acquisition, the data storage was not yet prepared to store diaPASEF. As such, the analysis.tdf file looks like a ddaPASEF sample in all regards. There are three noticable exception though:
* the `Precursors` table in the analysis.tdf is empty.
* the `Precursor` column of the `PasefFrameMsMsInfo` shows Null values since this could not be determined when saving the acquired data.
* fragment frames are indicated with a `MsMsType` of 8 (indeed indicating ddaPASEF), instead of a 9 (indicating diaPASEF in newer files, which eventually lead to several `DiaFrameMsMs...` tables instead of a `PasefFrameMsMs` table).

A fix is to update the `alphatims.bruker` source code with the following:

```
#>>> original lines 1616-1618
        isolation_mzs = self.fragment_frames.IsolationMz.values
        isolation_widths = self.fragment_frames.IsolationWidth.values
        precursors = self.fragment_frames.Precursor.values
#<<< copy paste and insert 10 lines below
        if (precursors[0] is None):
            frame_groups = self.frames.MsMsType.values[1:]  # exc. zeroth frame
            precursor_frames = np.flatnonzero(frame_groups == 0)
            group_sizes = np.diff(precursor_frames)
            group_size = group_sizes[0]
            if np.any(group_sizes != group_size):
                raise ValueError("Sample type not understood")
            precursors = (1 + frame_ids - frame_ids[0]) % group_size
            self.fragment_frames.Precursor = precursors
            self._acquisition_mode = "diaPASEF"
#>>> original lines 1619-1621, which now become lines 1629-1631
        scan_max_index = self.scan_max_index
        frame_max_index = self.frame_max_index
        quad_indptr = [0]
```

I will incorporate this in version 0.2.8, but since this is a bug which should not affect many other users and you have a developer version installed anyways I will not release this new version just yet.

Best,
Sander

flmeier, 2021-05-29:
Dear Sander, 
Thank you very much for your prompt response! The few extra lines fixed the problem!

`2021-05-29 23:23:28> Succesfully imported data from 20190319_tims01_FlMe_SA_BSA_45min_diaPASEF_C1_01_5714.d
`

Thanks a lot
Florian

closed 2021-05-29 by flmeier