MannLabs/alphatims #145: Discrepancy of m/z values between AlphaTims, OpenTIMS, and Bruker SDK

DarylWM, 2021-08-02:
Thank you for sharing this library. Its performance and slicing is really nice work.

**Describe the bug**

It seems too soon to call it a bug; it may well be something I'm doing. I've noticed that m/z values are different for the same raw points between AlphaTims, OpenTIMS, and Bruker SDK. The m/z values from OpenTIMS and the Bruker SDK are the same, but I've seen 5-8 ppm difference in m/z values from AlphaTims.

**To Reproduce**

Using AlphaTims:

```
file_name = './P3856_YHE211_1_Slot1-1_1_5104.d'
data = alphatims.bruker.TimsTOF(file_name)
df = data[
    {
        "frame_indices": 14612,
        "scan_indices": 33,
    }
]
df[['frame_indices','rt_values','scan_indices','mz_values','intensity_values']]

```

Using OpenTIMS:

```
path = pathlib.Path(file_name)
D = OpenTIMS(path)
df_l = []
for idx,d in enumerate(D.query_iter(D.ms1_frames, columns=('frame','mz','scan','intensity','retention_time'))):
    if ((d['retention_time'][0] >= rt_lower) and (d['retention_time'][0] <= rt_upper)):
        d['frame'] = d['frame'].astype(np.uint16, copy=False)
        d['mz'] = d['mz'].astype(np.float32, copy=False)
        d['scan'] = d['scan'].astype(np.uint16, copy=False)
        d['intensity'] = d['intensity'].astype(np.uint16, copy=False)
        d['retention_time'] = d['retention_time'].astype(np.float32, copy=False)
        df_l.append(pd.DataFrame(d))
df = pd.concat(df_l, axis=0, sort=False, ignore_index=True)
df[(df.scan==33) & (df.frame==14612)][['frame','retention_time','scan','mz','intensity']]

```

Using Bruker SDK:

```
td = timsdata.TimsData(file_name)
frame_points = []
frame_id = 14612
for scan_idx,scan in enumerate(td.readScans(frame_id=frame_id, scan_begin=0, scan_end=number_of_scans)):
    index = np.array(scan[0], dtype=np.float64)
    mz_values = td.indexToMz(frame_id, index)
    intensity_values = scan[1]
    scan_number = scan_idx
    number_of_points_on_scan = len(mz_values)
    for i in range(0, number_of_points_on_scan):
        mz_value = float(mz_values[i])
        intensity = int(intensity_values[i])
        d = {'frame_id':frame_id, 'mz':mz_value, 'scan':scan_number, 'intensity':intensity, 'retention_time_secs':retention_time_secs}
        frame_points.append(d)
df = pd.DataFrame(frame_points)
df[(df.scan==33) & (df.frame_id==14612)][['frame','retention_time_secs','scan','mz','intensity']]

```

**Expected behavior**

I don't know which framework is giving the correct m/z readings, but it seems odd they are different.

**Logs**

```
2021-08-02 15:07:21> Importing data from /media/big-ssd/experiments/P3856/raw-databases/P3856_YHE211_1_Slot1-1_1_5104.d
2021-08-02 15:07:21> Reading frame metadata for /media/big-ssd/experiments/P3856/raw-databases/P3856_YHE211_1_Slot1-1_1_5104.d
2021-08-02 15:07:22> Reading 25,701 frames with 1,292,274,220 detector strikes for /media/big-ssd/experiments/P3856/raw-databases/P3856_YHE211_1_Slot1-1_1_5104.d
2021-08-02 15:07:44> Indexing /media/big-ssd/experiments/P3856/raw-databases/P3856_YHE211_1_Slot1-1_1_5104.d...
2021-08-02 15:07:44> Opening handle for /media/big-ssd/experiments/P3856/raw-databases/P3856_YHE211_1_Slot1-1_1_5104.d
2021-08-02 15:07:44> Fetching mobility values from /media/big-ssd/experiments/P3856/raw-databases/P3856_YHE211_1_Slot1-1_1_5104.d
2021-08-02 15:07:44> Closing handle for /media/big-ssd/experiments/P3856/raw-databases/P3856_YHE211_1_Slot1-1_1_5104.d
2021-08-02 15:07:44> Opening handle for /media/big-ssd/experiments/P3856/raw-databases/P3856_YHE211_1_Slot1-1_1_5104.d
2021-08-02 15:07:44> Fetching mz values from /media/big-ssd/experiments/P3856/raw-databases/P3856_YHE211_1_Slot1-1_1_5104.d
2021-08-02 15:07:44> Closing handle for /media/big-ssd/experiments/P3856/raw-databases/P3856_YHE211_1_Slot1-1_1_5104.d
2021-08-02 15:07:46> Succesfully imported data from /media/big-ssd/experiments/P3856/raw-databases/P3856_YHE211_1_Slot1-1_1_5104.d

```

**Screenshots**

[image]

**Version (please complete the following information):**

I installed with pip in a fresh conda environment:
```
conda create --name alphatims python=3.8
conda activate alphatims
pip install "alphatims[stable]"

```

```
2021-08-02 15:07:21> Platform information:
2021-08-02 15:07:21> system        - Linux
2021-08-02 15:07:21> release       - 4.15.0-147-generic
2021-08-02 15:07:21> version       - #151-Ubuntu SMP Fri Jun 18 19:21:19 UTC 2021
2021-08-02 15:07:21> machine       - x86_64
2021-08-02 15:07:21> processor     - x86_64
2021-08-02 15:07:21> cpu count     - 12
2021-08-02 15:07:21> cpu frequency - 3331.27 Mhz
2021-08-02 15:07:21> ram           - 51.4/62.8 Gb (available/total)

```

```
2021-08-02 15:07:21> Python information:
2021-08-02 15:07:21> alphatims  - 0.2.8
2021-08-02 15:07:21> bokeh      - 
2021-08-02 15:07:21> click      - 8.0.1
2021-08-02 15:07:21> datashader - 
2021-08-02 15:07:21> h5py       - 3.1.0
2021-08-02 15:07:21> hvplot     - 
2021-08-02 15:07:21> numba      - 0.53.1
2021-08-02 15:07:21> pandas     - 1.3.1
2021-08-02 15:07:21> psutil     - 5.8.0
2021-08-02 15:07:21> python     - 3.8.5
2021-08-02 15:07:21> python-lzf - 
2021-08-02 15:07:21> pyzstd     - 0.14.4
2021-08-02 15:07:21> selenium   - 
2021-08-02 15:07:21> tqdm       - 4.61.1

```

**Additional context**

All three libraries are using a float64 for the m/z value. I was thinking perhaps there's an intermediate step in AlphaTims where the m/z values are downcast to 32 bits? Just a thought.

swillems, 2021-08-02:
Dear @DarylWM, we are pleased to hear you already enjoy AlphaTims.

You are right, there there indeed are can indeed be minor differnces between the different tools. This is not only true for the mz values, but potentially can also happen for mobility and intensity values (although these differences are often so minor that you probably will not be able to notice them at all). In brief, this is because the Bruker SDK actually retuns calibrated data. Since OpenTims utilizes the Bruker SDK under the hood, they naturally work with the same calibrated data and hence have the same results. However, AlphaTims uses the raw data directly, which in essence are only the TOF indices. Based on these TOF indices and the instument settings (lowest set mz value and largest set mz values), we estimate the mz values for each arrival time. This generally gives decent results and is one of the reasons of our performance boost. We are currently working on an improved calibration (assuming calibrant sprays are used), but this is not available just yet and I unfortunately cannot give you a precise timeline when it will be. If calibration is crucial for you, I can suggest a hybrid approach where you use AlphaTims to retrieve the raw data and combine this with the Bruker SDK directly:

```
file_name = './P3856_YHE211_1_Slot1-1_1_5104.d'
td = timsdata.TimsData(file_name)
data = alphatims.bruker.TimsTOF(file_name)
frame_id = 14612
df = data[frame_id, 33]
mz_values = td.indexToMz(frame_id, df['tof_indices'].values)
df[['frame_indices','rt_values','scan_indices','mz_values','intensity_values']]
```

Let us know if this helps!

DarylWM, 2021-08-02:
Thanks for the reply @swillems ; that makes sense.

closed 2021-08-02 by DarylWM

DarylWM, 2021-08-12:
Dear Sander. On Linux are you using the Bruker library to [convert from indices to m/z values](https://github.com/MannLabs/alphatims/blob/dc0a2ff78a3dfe0fe5e89959d7ec4beafa8bcbb9/alphatims/bruker.py#L44), and using it by default [here](https://github.com/MannLabs/alphatims/blob/dc0a2ff78a3dfe0fe5e89959d7ec4beafa8bcbb9/alphatims/bruker.py#L870)? In that case sorry I don't understand why the raw m/z values are different.

swillems, 2021-08-12:
Dear @DarylWM. Let me begin with thanking you very much for actually digging into the code a little deeper. I hope that the documentation is decent and that you are able to navigate your way through it without too many difficulties.

Now, to your question. Let me try to break it down in what happens:
1. As mentioned earlier, the raw data that is actually acquired with a TOF instrument are purely intensity values coupled to time-of-flight values. Based on the set voltage and length of a TOF tube, the TOF values can directly be converted to mz values. Since this follows a simple quadratic equation, we can also use the [lowest and highest set mz values](https://github.com/MannLabs/alphatims/blob/dc0a2ff78a3dfe0fe5e89959d7ec4beafa8bcbb9/alphatims/bruker.py#L985) to calculate mz values for each [TOF value](https://github.com/MannLabs/alphatims/blob/dc0a2ff78a3dfe0fe5e89959d7ec4beafa8bcbb9/alphatims/bruker.py#L1013) instead of relying on the voltage and TOF tube length.
2. Unfortunately, these settings as provided by Bruker are not always as [accurate](https://github.com/MannLabs/alphatims/blob/dc0a2ff78a3dfe0fe5e89959d7ec4beafa8bcbb9/alphatims/bruker.py#L31). I have seen samples that state the mz values are in a range of (100, 1700), whereas the actual lowest and highest mz values were actually closer to (95, 1705). This means that our mz estimation will yield utterly wrong results. Luckily, there is a Bruker DLL available on Windows and Linux as you correctly pointed out. By using this DLL, we do not need to estimate the conversion from TOF to mz, as we can just retrieve the complete [mz array](https://github.com/MannLabs/alphatims/blob/dc0a2ff78a3dfe0fe5e89959d7ec4beafa8bcbb9/alphatims/bruker.py#L1001). 
3. This brings us to the final complexity that hopefully gives you the answer you are looking for. As stated, the conversion of TOF values to mz values relies on the set voltage and length of the TOF tube. While these in theory should be constant, there are always minor fluctuations over time. Voltages for instance require a constant energy influx to be maintained, while the length of a TOF tube can actually vary slightly due to shrinkage or expansion as an effect of temperature changes. Even though these might seem like minimal effects, high-res MS instruments are accurate enough to detect these changes. This effectively means that the the same TOF value in the beginning of a run does not necessarily equate to the same mz value in the end of a run. This is actually quite clear when using the Bruker DLL, since the `mz_values = td.indexToMz(frame_id, index)` explicitly requires a `frame_id`. Indeed, when you call the same function with a different `frame_id`, you should be able to see that equal TOF values do not always yield equal mz values. Note btw, that you can provide any TOF index, it does not matter if this TOF index was actually measured in the specific frame or not.

In summary, TOF-to-mz conversion is not constant throughout a run. Bruker calibrates indivudual frames, but at the moment AlphaTims assumes a constant TOF-to-mz relation. For the next AlphaTims release, we have implemented a global calibrabration based on calibrant sprays that should solve the issues of step 2 without relying on the Bruker DLL aymore (making it truly cross-platform). There are future plans to also solve step 3, but at the timing of writing this is still just a TODO item. I hope this helped. If you are interested in the matter, I would happily collaborate directly to try and implement this functionality!

DarylWM, 2021-08-13:
Dear Sander. Thank you for the explanation. I'm keen to help solve these problems if I can. Would enabling the Discussion feature on the repo be the best way to continue the discussion?

Regards,
Daryl.

swillems, 2021-08-13:
That sounds great! I haven't used the discussion feature of GitHub before. I have just enabled it and am curious to see how it works.

swillems, 2021-11-04:
@DarylWM , it has been a while since we discussed this calibration issue. I recently read your TFDE preprint (by now AlphaTims is also officially [published](https://doi.org/10.1016/j.mcpro.2021.100149) as well) and saw you already do a great deal of processing there. While AlphaTims is primarily intended for raw data accession and visualization, I still would consider incorporating some of your ideas directly in AlphaTims. As you previously suggested to use the discussion option, I was wondering if you still had an active interest in improving this.

DarylWM, 2021-11-07:
Sander, congratulations on the publication of your paper as well. TFDE uses AlphaTims for access and storage of raw data; it's been a big help to me. It turns out the mass discrepency problem I originally raised is resolved in TFDE's mass recalibration step quite well.

Discussing what to incorporate from TFDE's spectra simplification steps into AlphaTims and how to do it would be an interesting discussion, so yes I'd like to join in.

swillems, 2021-11-08:
Dear Daryl, 
That's great to hear! I think this should make for a great collaboration and I have opened a discussion (#157) to get started with this and suggest we move over there for transparancy.

btcooooper, 2022-07-18:
I know this is closed, but I see a consistent discrepancy between the raw tof indices displayed by AlphaTims and by TimsR. For the data file I've been inspecting, the tof indices shown by AlphaTims are consistently one greater than those shown by TimsR.

swillems, 2022-07-19:
Do you have some examples? A one-off like that is a typical bug, but I am surprised this only surfaces now. Perhaps naive question, but can this be the difference between 1-indexed or 0-indexed programming languages?

swillems, 2022-07-21:
Hmm, very interesting. Do you have some more information about the file. In particular, I would be interested to know which "TimsCompressionType" is used (open the ...d/analysis.tdf with e.g. [DB Browser for SQLite](https://sqlitebrowser.org/dl/) and browse the "GlobalMetadata" table to find it).

btcooooper, 2022-07-21:
TimsCompressionType = 2