MannLabs/alphatims #161: recode polarity to int

liquidcarbon, 2022-02-01:
Hello,

You provide the `drop_polarity` argument to `read_bruker_sql` but do not use it.  The code says:
```
        frames = pd.DataFrame(
            {
                col: pd.to_numeric(
                    frames[col]
                ) for col in frames if col != "Polarity"
            }
        )
```
There is probably an `if drop_polarity:` missing.
https://github.com/MannLabs/alphatims/blob/master/alphatims/bruker.py#L249

I understand the motivation for keeping only numeric types.  Personally, I would prefer the polarity was never dropped.
- there is no place to infer polarity from.  I wish it was in the global metadata but it's not.
- how about keeping polarity but recoding it into `-1` and `+1`?  I tried opening a pull request but had trouble with pushing my commit here.

`frames['Polarity'] = frames['Polarity'].apply(lambda x: int(x+'1'))` then all your dtypes are numeric.

swillems, 2022-02-02:
Dear Alexander,

Thanks for looking into this. You are right that this is not properly implemented. It now is available on the [development branch](https://github.com/MannLabs/alphatims/pull/162/files?file-filters%5B%5D=.py&show-viewed-files=true). I have also made the option available to trigger this directly through the `alphatims.bruker.TimsTOF` constructor. Even though your solution is more elegant, I have not made it the default to maintain backward compatability. This might change in the future

swillems, 2022-03-30:
This is in master branch by now. If a problem remains, feel free to open up a new issue.

closed 2022-03-30 by swillems