MannLabs/alphatims #191: convert_from_indices

liquidcarbon, 2022-04-27:
Greetings!

https://github.com/MannLabs/alphatims/blob/f6e55e899b5a92e831852d7b9d96a2cc00a685eb/alphatims/bruker.py#L1262
I'd like to access the first X frames of the data.  I don't want to go with standard indexing because I only need to access a few of the fields.  I'm attempting to use `.convert_from_indices` and it produces strange results.

```
# accessing the first two frames
data[[1,2]].shape
# returns (6879, 14)

data.convert_from_indices(
    raw_indices=None,
    frame_indices=[1,2],
    return_tof_indices=True,
)

_['tof_indices'].shape

# returns (1, 8838754)
# an array of arrays!
# the length of inner array is the length of whole run
```

I've read the function 5 times but don't understand how this happens... :)

liquidcarbon, 2022-04-27:
I can get around it by asking
```
data.convert_from_indices(
    raw_indices=data[[1,2], "raw"],
    return_tof_indices=True,
)
```

Not a big deal - was just wondering if that's expected behaviour.

swillems, 2022-04-28:
Dear @liquidcarbon, thanks for checking in again. Your second approach with "raw" slicing is the route I always take myself as well.

That said, I do understand the confusion with the first approach. From a technical perspective, it boils down to lines https://github.com/MannLabs/alphatims/blob/f6e55e899b5a92e831852d7b9d96a2cc00a685eb/alphatims/bruker.py#L1412-L1413 

Since `return_tof_indices` is True, AlphaTims will slice the `tof_indices` as requested. However, providing a `None` slice is the same as a full `:` or Ellipsis `...` slice, hence all elements are returned.

The confusion probably lies more in the conceptual idea of the function. It is intended to convert `raw_indices` to their actual coordinates. The additional `frame_indices` and other index options are only intended for performance reasons and to avoid additional computation. 

This is somewhat hinted at by the fact that `raw_indices` are the only accepted non-keyword argument. Probably it is a good idea to remove the default `None` option to show this is an essential keyword and to update some of the documentation. If you have any suggestions, feel free to make PR into the dev branch. If not, I'll probably get to this myself in the future at some point...

closed 2022-10-14 by sander-willems-bruker