MannLabs/alphatims #202: No plots are shows when using the GUI in Google Chrome or Firefox

MarcIsak, 2022-06-07:
**Describe the bug**
I have created an 'alphatims docker image' (see attachment) for alphatims 0.3.2 that I run in WSL2,  and I can manage to port the gui to my browser (localhost:4500). However, once I have uploaded a *.d file, there are no plots displayed. I can't see any error messages being displayed either (see attached log)

**Dockerfile to create  the alphatims image (zipped)**

[Dockerfile.zip](https://github.com/MannLabs/alphatims/files/8853472/Dockerfile.zip)

**Setup file run by CMD in dockerfile**
[setup.zip](https://github.com/MannLabs/alphatims/files/8853499/setup.zip)

**To Reproduce**
Steps to reproduce the behavior (more detailed in the Dockerfile):
1. 'conda create -n alphatims python=3.8'
2. 'conda install -n alphatims pip'
3. 'conda activate alphatims'
4. pip install 'alphatims[plotting]'
5. Run alphatims gui --port:4500
6. Open browser and load http://localhost:4500
7. Load either *.d folder or *.hdf file

**Expected behavior**
Expect plots to appear once the file is uploaded in the gui. The plots are compressed and do not show anything. 

**Logs**
Please provide the log (see the AlphaTims terminal on where to find it).
[log_20220607125907.txt](https://github.com/MannLabs/alphatims/files/8853436/log_20220607125907.txt)

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Version (please complete the following information):**
 - Installation Type: pip
 - If no log is available, provide the following:
   - Platform information
     - system    Linux
     - release   5.10.102.1-microsoft-standard-WSL2
     - version   #1 SMP Wed Mar 2 00:30:59 UTC 2022
     - machine   x86_64
     - processor [e.g. i386]
     - cpu count 20
   - Python information:
     - alphatims version 0.3.2
     - alphatims  - 0.3.2
     - bokeh      - 2.4.3
     - click      - 8.1.3
     - datashader - 0.14.0
     - h5py       - 3.7.0
     - hvplot     - 0.8.0
     - numba      - 0.55.2
     - pandas     - 1.4.2
     - psutil     - 5.9.1
     - python     - 3.8.13
     - python-lzf -
     - pyzstd     - 0.15.2
     - selenium   - 4.2.0
     - tqdm       - 4.64.0

**Additional context**
Add any other context about the problem here. Attached log files or upload data files if possible.

swillems, 2022-06-07:
Thanks for the feedback. Can you try to update step 4 to: `pip install 'alphatims[stable,plotting-stable]'` and see if this works better? (You can also update the current environment with an `--upgrade` flag after the pip install ...)?

swillems, 2022-06-09:
Hey @MarcIsak . Did you get it working? If yes, I will close this issue. If not, you can try with the latest release (v1.0.0), although you should still use the `stable` installation (this is now explicitly mentioned in the readme as well).

closed 2022-08-16 by swillems