MannLabs/alphatims #218: python 3.10

liquidcarbon, 2022-08-25:
Hello!  I cloned and edited `alphatims/__init__.py` to allow for python 3.10 and built a wheel with it.  It appears to install and work fine.  Would you mind officially allowing python 3.10?

Some key dependencies:
```
  • Installing llvmlite (0.39.0)
  • Installing numpy (1.22.4)
  • Installing numba (0.56.0)
  • Installing alphatims (1.0.0 alphatims/alphatims-1.0.0-py3-none-any.whl)
```

liquidcarbon, 2022-10-03:
have you had a chance to look into allowing python 3.10?

sander-willems-bruker, 2022-10-14:
Sorry, seems like I stopped receiving mails from the AlphaTims repo. I updated the versioning to: `__python_version__ = ">=3.8,<4"`, on the dev branch, and will probably make an official new release next week. Thanks for checking this!

closed 2022-10-14 by sander-willems-bruker

jspaezp, 2023-01-19:
Hello there! I wanted to ask if this is officially supported? (if so, I think the readme could be updated to specify the versions allowed!)

Sections like `alphaTims can be installed in an existing Python 3.8 environment` might be misleading.

Let me know if you would like PR to fix that!
Kindest wishes,
Sebastian

sander-willems-bruker, 2023-01-20:
Yes, that is official by now! Mainting docs is always difficult though, feel free to make a PR into the dev branch and replace any of the docs which are faulty (perhaps also the tutorial nbs)! Thanks for pointing it out.