MannLabs/alphatims #242: 1.0.5 GUI wont start

kronigert, 2022-11-10:
Hello,

I just stumpled across the following error: I installed version 1.0.5. of AlphaTIMS via the GUI installer. When I try to open AlphaTIMS with the GUI it wont open (tried it with several browsers set as default). I put the cmd-log of the error message of the quitting below. As this error does not occur with version 1.0.4. of AlphaTIMS I dont think something is wrong with my setup.

Thank you for the cool software.
All the best,
Tobias

**Logs**
Traceback (most recent call last):
  File "alphatims_pyinstaller.py", line 14, in <module>
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "PyInstaller\loader\pyimod03_importers.py", line 495, in exec_module
  File "alphatims\gui.py", line 15, in <module>
    import alphatims.bruker
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "PyInstaller\loader\pyimod03_importers.py", line 495, in exec_module
  File "alphatims\bruker.py", line 277, in <module>
    @alphatims.utils.njit(nogil=True)
  File "alphatims\utils.py", line 444, in njit
    import numba
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "PyInstaller\loader\pyimod03_importers.py", line 495, in exec_module
  File "numba\__init__.py", line 190, in <module>
    _ensure_critical_deps()
  File "numba\__init__.py", line 133, in _ensure_critical_deps
    import scipy
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "PyInstaller\loader\pyimod03_importers.py", line 495, in exec_module
  File "scipy\__init__.py", line 76, in <module>
    _delvewheel_init_patch_1_0_1()
  File "scipy\__init__.py", line 67, in _delvewheel_init_patch_1_0_1
    os.add_dll_directory(libs_dir)
  File "os.py", line 1109, in add_dll_directory
FileNotFoundError: [WinError 2] Das System kann die angegebene Datei nicht finden: 'C:\\Users\\XXX.XXX\AppData\\Local\\Programs\\AlphaTims\\scipy.libs'
[20160] Failed to execute script 'alphatims_pyinstaller' due to unhandled exception!

adang1345, 2022-11-18:
To the developer: This is a bug in PyInstaller that was fixed in version 5.6 (https://github.com/pyinstaller/pyinstaller/pull/7170). It looks like that needs to be updated here: https://github.com/MannLabs/alphatims/blob/0f1275aaa779dc7c7e3dbf2707f14356b08c9e4b/misc/one_click_windows/create_installer_windows.bat#L11

sander-willems-bruker, 2022-11-21:
Hi @kronigert . Thanks for the report. Thanks to @adang1345 's suggestion, the latest pyinstaller is now used, whihc fixes the issue in v1.0.6.

closed 2022-11-21 by sander-willems-bruker