MannLabs/alphatims #256: [Pitch] Docker Image

jspaezp, 2023-02-02:
Hello there! I was wondering if you would be interested in adding a docker image to the release cycle of the package.

I would gladly make PR for it if you want!

Thanks for the good work on the package :)
-Sebastian

sander-willems-bruker, 2023-02-03:
Hi @jspaezp , thanks for the kind words in interest. I have little to no experience with docker myself, but agree that it might be a valuable addition. Feel free to make a PR and we'll try and integrate it as well as possible. If it works out smoothly, I would be happy to collaborate actively and try to build some sort of template around it that might be useful for other packages in the AlphaPept ecosystem.

jspaezp, 2023-02-03:
Hello @sander-willems-bruker ! that is great to hear, in my experience adding the docker file is fairly smooth for python packages (assuming one will use the GitHub container registry) ... 

The docker file just uses the built distribution that would be pushed to pypi
https://github.com/wfondrie/mokapot/blob/master/Dockerfile (example in a package that I am a co-maintainer in, that also uses conda as the main dependency and env manager)

And on the CI-CD side of things it just needs the addition of a gh action for it...
Mokapot uses as separate workflow:
https://github.com/wfondrie/mokapot/blob/master/.github/workflows/docker.yml

I have used my same deployment workflow (since I can just re-use the wheel build).

I think I could have a draft of the PR by Monday, LMK what branch you want me to use as a base for it :)

best,
Sebastian

sander-willems-bruker, 2023-02-06:
Sounds great! If you branch of development, we can merge it in there. Thanks again for the support.

jspaezp, 2023-04-04:
Hello there! @sander-willems-bruker I have submitted a PR (#268), lmk if the scope is what you feel would be enough or if you have any additional requirements (more documentation/change CICD/permissions/additional things to bundle in the image....)

Best!
Sebastian

sander-willems-bruker, 2023-07-18:
Finally got to merging this in develop. Thanks for the help!

closed 2023-07-18 by sander-willems-bruker

jspaezp, 2023-07-27:
https://github.com/MannLabs/alphatims/pull/268#issuecomment-1648207397

posting here for context!
Permissions required for a public image.

sander-willems-bruker, 2023-07-28:
Yep, turns out this requires admin at organization level, I forwarded to the people that can enable this so hopefully will be integrated soon. If not, ping me;)

swillems, 2023-09-22:
Dear @jspaezp . It took a little longer than expected, but the docker image should be available by now. Feel free to test it out and let me know what does and  does not work!

jspaezp, 2023-09-24:
Thanks a lot @swillems ! this is great!