MannLabs/alphatims #265: [Question] precursor_indices definition

jspaezp, 2023-02-27:
Hello!

I have been exploring the package and had a question on how the 'precursor_indices' is defined.
When I filter based on the precursor index (1 in this case), get two distinct quad isolation.

This is contrary to what I thought would be the case (I was expecting each precursor to be only one iso window), so I wanted to know whether this is an expected behavior or how the precursor index is being defined.

Is there any indexing to access unique quad isolation windows in a sequential manner?

Appreciating your help!
-Sebastian

Snippet showing how I am getting the results

```python
# This file comes from the PRIDE repo for the alphatims paper
# curl ftp.pride.ebi.ac.uk/pride/data/archive/2022/02/PXD028735/LFQ_timsTOFPro_diaPASEF_Ecoli_01.d.zip --output ../data/ecoli_timsTOFPro_diaPASEF.d.zip
dia_data = TimsTOF("../data/LFQ_timsTOFPro_diaPASEF_Ecoli_01.d", mmap_detector_events=False)
inds = dia_data[{"precursor_indices": 1}, "raw"]
out = dia_data.convert_from_indices(inds,
        raw_indices_sorted=True,
        return_quad_mz_values=True,
        )
np.unique(out["quad_low_mz_values"], return_counts=True)
# (array([400., 800.]), array([ 20845610, 119548979]))
```

sander-willems-bruker, 2023-02-28:
Dear @jspaezp . This is the original intended behavior (see page 18-19 of the [GUI manual](https://github.com/MannLabs/alphatims/blob/master/alphatims/docs/gui_manual.pdf)).
I understand that this indeed can be a confusing situation, but at the time I felt it was the best option since "precursor_indices" is first-and-foremost a DDA feature (if only by using the name "precursor") and only an afterthought for DIA. After the initial implementation two years ago, I now agree that defining DIA precursor_indices as a single "window" might make more sense than the current definition of a "window group". That said, with new acquisitions such as [synchroPASEF](https://www.mcponline.org/article/S1535-9476(22)00297-3/fulltext), the "window group" definition might again become the more dominant use case. As such, I will leave it unchanged for now to ensure backwards compatibility and avoid any further confusion.

With respect to your second question on how to select unique quad isolition windows in a sequential manner, I personally tend to use quad setting directly to do this. Notice that you should be able to find all those easily by viewing the analysis.tdf with any sql browser, or by checking out the .cycle attribute of a TimsTOF object.

jspaezp, 2023-02-28:
@sander-willems-bruker Thank you very much for the reply! that makes a lot of sense now!

closed 2025-03-06 by mschwoer