MannLabs/alphatims #269: Synchro/slice data import

videlc, 2023-04-05:
Hi,

Thanks a lot for developing alphatims.
I am currently facing an issue when importing a segmented (metabolomics) synchro/slicePASEF analysis, yielding in the following error:
```
File /usr/miniconda3/envs/py3.8/lib/python3.8/site-packages/alphatims/bruker.py:2306, in TimsTOF.set_cycle(self)
   2304     low_mz = row.IsolationMz - row.IsolationWidth / 2
   2305     high_mz = row.IsolationMz + row.IsolationWidth / 2
-> 2306     cycle[
   2307         frame,
   2308         scan_begin: scan_end,
   2309     ] = (low_mz, high_mz)
   2310     precursor_frames[frame] = False
   2312 cycle[precursor_frames] = (-1, -1)

IndexError: index 166 is out of bounds for axis 0 with size 4
```

Let me know if any need further information is needed or a minimal data example via email.
Best regards,
Vivian

sander-willems-bruker, 2023-07-18:
Dear Vivian, would it be possible for you to share the data? Otherwise it is very difficult to track down what goes wrong in this particular case.

videlc, 2023-07-18:
Dear Sander,
Test file and example jupyter notebook have been sent to you via email minutes ago. 
Thanks,
Vivian

sander-willems-bruker, 2023-07-18:
Dear Vivia, thanks for sharing the test file via a private message. It turns out this is particular sample actually defines two acquisition modes, i.e. 0 and 9. In effect, the first 166 are defined as MS1 frames and only as of frame 167 does a cycle DIA acquistion start. This confuses alphatims, as it assumes a single cample only can have one acquistion mode. I would be curious to know how you set up this acquisition.

videlc, 2023-07-18:
Default metabolomics (with VIPHESI)  methods incoporate MS1 scans at the beginning of the run which is timed with the tims 6 way valve. Valve switches at the beginning of the run to ensure that a "plug" of calibrant is being pushed into the ion source with ensures mz + IM recalibration for every run (see capture with salt cluster example).
You can load default metabolomics pos in timscontrol to get an example method (4D_metabolomics_pos).
Please note that early-run recalibration is standard in metabolomics in many labs, so I might not be the only one interested into a fix of this issue. 

[image]

sander-willems-bruker, 2023-07-25:
This is not a trivial fix I am afraid, best I might be able to do at short notice is a bandaid and even that is not straightforward. Do you use AlphaTims as a gui, cli or python package?

videlc, 2023-07-25:
Hi,
Python+Jupyter only.
Note that I was not aware of the issue while making the methods. Since this, I've unfortunately deleted the recalibration segment so I'm able to work with alphatims. Of course, being able to perform a pre-run calilbration is a highly valuable feature that other users (metabolomics and proteomics) are going to request later IMO.
Best regards,
Vivian

closed 2025-03-13 by sander-willems-bruker