MannLabs/alphatims #273: [Maintainance] CICD

jspaezp, 2023-07-24:
While working on the CICD of the project to publish the docker image I noticed a couple of things that you/we might want to consider addressing

1. On the `Version_Bumped ` stage of the CICD it throws this warning
```
Warning: The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
```
This could be updated.

2. Version bumping is handled in a very ... bespoke way using sed, a tool like `bumpver` could be used to make this automatic.
3. A lot of the packaging an versioning could be simplified by migrating to the `pyproject.toml` structure (and would at the same time remove many redundant places where configurations and dependencies are stablished).
4. There are maguey processes spread in multiple scripts around the repo. It would be a good idea to centralize those in a MAKEFILE (even ion the makefile just calls those scripts).

lmk what you think
Best
Sebastian

sander-willems-bruker, 2023-07-24:
Thanks for the feedback! Great to see people are actively contributing to AlphaTims, much appreciated!
I am a fan of the [boyscout rule](https://levelup.gitconnected.com/the-boy-scouting-rule-a-simple-yet-effective-programming-principle-5d504c7dd993#:~:text=The%20Boy%20Scouting%20Rule%20is,environment%20cleaner%20than%20they%20found), feel free to modify things that can be improved and make a PR for them.

To answer your questions:

1. Yes, this should indeed be fixed! I modified the workflow accordingly.
2. This is hacky to test it only. Actual updating is done via a manual `bumpversion` command (see misc/bumpversion.cfg). With this check we are currently sure that there is no duplicaiton of versions, as the workflow will fail if it is not updated with respect to pypi. I would be happy to accept a cleaner check if you have one!
3. I generally agree. However, I have also found that some of the variables set in `pyproject.toml` and `settings.cfg` files are useful to have present in the actual source code for e.g. cli prompts. As such, the central place for these in AlphaTims is the `__init__.py` file. The only value that has to be updated in vase something changes is the version, which is controlled by a single bumpversion command.
4. Never worked with MAKEFILEs, happy to get a crash course or example of how you envision this is in AlphaTims!

jspaezp, 2023-07-27:
1. That was fast! Good job!
2. Ohhh that is interesting, I understand the rationale! (note: https://github.com/peritus/bumpversion `bumpversion` seems abandoned)
3. Oh ok, I think we share rationale but differ in approach, I state everything in the pyproject.toml and pull from it in my __init__.py (https://github.com/TalusBio/flimsay/blob/release/flimsay/__init__.py for example).
4. https://makefiletutorial.com this guide is good. If I have time sometime soon I might make a small draft of how it could be used.

Thanks a lot for the replies!

sander-willems-bruker, 2023-07-28:
1. Was a very minimal change;)
2. Thanks for pointing out the deprecation. At closer sight though, it looks like a maintained fork is already used as a drop-in replacement: https://github.com/c4urself/bump2version/#installation.
3.  That's a nice take on it. I'll definately try that out in the future as I agree that a .toml is a more standardized/accepted way in the community and probably more flexible.
4. I'll check that out, but probably won't get to anytime soon, let alone implementing it...

closed 2023-07-28 by sander-willems-bruker