MannLabs/alphatims #275: Exporting diaPASEF as mgf

lazear, 2023-08-02:
**Describe the bug**
Exporting diaPASEF data to mgf fails - but hdf5 succeeds. I can successfully export ddaPASEF files to mgf.

**To Reproduce**
```
$ alphatims export mgf HT_20230705_Belharra_200SPD_Exp1_Yeast_300ng_blank_A1_S1-A1_1_2636.d
2023-08-02 12:10:31> File HT_20230705_Belharra_200SPD_Exp1_Yeast_300ng_blank_A1_S1-A1_1_2636.d is not a ddaPASEF file, nothing to do.
```

```
$ alphatims export hdf HT_20230705_Belharra_200SPD_Exp1_Yeast_300ng_blank_A1_S1-A1_1_2636.d
2023-08-02 12:09:36> Successfully wrote TimsTOF data to HT_20230705_Belharra_200SPD_Exp1_Yeast_300ng_blank_A1_S1-A1_1_2636.hdf.
```
**Expected behavior**
Export to MGF (I'm trying to create mzMLs)

**Logs**
```
*******************
* AlphaTims 1.0.8 *
*******************
2023-08-02 12:10:28> Platform information:
2023-08-02 12:10:28> system        - Windows
2023-08-02 12:10:28> release       - 10
2023-08-02 12:10:28> version       - 10.0.22621
2023-08-02 12:10:28> machine       - AMD64
2023-08-02 12:10:28> processor     - Intel64 Family 6 Model 151 Stepping 2, GenuineIntel
2023-08-02 12:10:28> cpu count     - 20
2023-08-02 12:10:28> cpu frequency - 3600.00 Mhz
2023-08-02 12:10:28> ram           - 16.2/31.9 Gb (available/total)
2023-08-02 12:10:28>
2023-08-02 12:10:28> Python information:
2023-08-02 12:10:28> alphatims  - 1.0.8
2023-08-02 12:10:28> bokeh      -
2023-08-02 12:10:28> click      - 8.1.6
2023-08-02 12:10:28> datashader -
2023-08-02 12:10:28> h5py       - 3.9.0
2023-08-02 12:10:28> hvplot     -
2023-08-02 12:10:28> jinja2     -
2023-08-02 12:10:28> numba      - 0.57.1
2023-08-02 12:10:28> pandas     - 2.0.3
2023-08-02 12:10:28> psutil     - 5.9.5
2023-08-02 12:10:28> python     - 3.10.8
2023-08-02 12:10:28> python-lzf -
2023-08-02 12:10:28> pyzstd     - 0.15.9
2023-08-02 12:10:28> selenium   -
2023-08-02 12:10:28> tqdm       - 4.65.0
2023-08-02 12:10:28>
2023-08-02 12:10:28> Current AlphaTims version is up-to-date with GitHub.
2023-08-02 12:10:28>
2023-08-02 12:10:28> Running CLI command `alphatims export mgf` with parameters:
2023-08-02 12:10:28> bruker_raw_data             - HT_20230705_Belharra_200SPD_Exp1_Yeast_300ng_blank_A1_S1-A1_1_2636.d
2023-08-02 12:10:28> centroiding_window          - 5
2023-08-02 12:10:28> disable_log_stream          - False
2023-08-02 12:10:28> disable_overwrite           - False
2023-08-02 12:10:28> keep_n_most_abundant_peaks  - -1
2023-08-02 12:10:28> log_file                    - D:\Github\scratch\bruker\.env\lib\site-packages\alphatims\logs\log_20230802121028.txt
2023-08-02 12:10:28> threads                     - 19
2023-08-02 12:10:28>
2023-08-02 12:10:29> Importing data from HT_20230705_Belharra_200SPD_Exp1_Yeast_300ng_blank_A1_S1-A1_1_2636.d
2023-08-02 12:10:29> Using HDF import for HT_20230705_Belharra_200SPD_Exp1_Yeast_300ng_blank_A1_S1-A1_1_2636.hdf
2023-08-02 12:10:31> Successfully imported data from HT_20230705_Belharra_200SPD_Exp1_Yeast_300ng_blank_A1_S1-A1_1_2636.d
2023-08-02 12:10:31> File HT_20230705_Belharra_200SPD_Exp1_Yeast_300ng_blank_A1_S1-A1_1_2636.d is not a ddaPASEF file, nothing to do.
2023-08-02 12:10:31> Analysis done in 2.83 seconds.
2023-08-02 12:10:31> WARNING: Temp mmap arrays were written to C:\Users\mlazear\AppData\Local\Temp\temp_mmap_zk1n8ojb. Cleanup of this folder is OS dependant, and might need to be triggered manually! Current space: 174,547,656,704
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (please complete the following information):**
 - Installation Type [e.g. One-Click Installer / Pip / Developer]: `pip install alphatims` in a fresh python venv

**Additional context**

Happy to transfer the file if needed

sander-willems-bruker, 2023-08-30:
Dear @lazear . Cliche as it gets, this is a typical "It's not a bug, it's a feature" situation (see the message itself `not a ddaPASEF file, nothing to do.`). The mgf and hdf export option serve two very different purposes:
* **hdf**: This format is intended to capture the full raw data, including the indexing. Please note that this is a custom format for AlphaTims (described in the [paper](https://www.mcponline.org/article/S1535-9476(21)00121-3/fulltext)). This allows to:
  * Speed up future data accession.
  * Make the data more transparant than the native .d folder (specifically the .tdf_bin file)
  * Pass around data more easily. While OS independent, AlphaTims is able to take minimal advantage of the Bruker SDK to improve conversion from tof indices to mz values and scan indices to im values on Windows/Linux. By exporting an HDF, this improved conversion can also be utilized on all other OS, including macOS.
* **mgf**: Many tools cannot directly use a ddaPASEF data for (proteomic) identification. The primary reason is that spectra in PASEF data are not fully comparable to traditional spectra due to the im separation. The way to make this compatible, is by aggregating PASEF "spectra" from multiple scans (thus with multiple different im values) and projecting these on the mz axis to obtain a traditional MS2 spectrum that purely has mz and intensity values. In DIA, this is not properly defined as windows are not directly related to precursors and aggregation of multiple PASEF scans can be done in any way the user wants. The alternative would be to create mgf files that skip data aggregation all together (this would also allow MS1 scans), but this leads to huge files which are generally cumbersome to work with.

I hope this makes sense. If you want to convert .d directly to mzml files, I think that is actually possible with [msconvert](https://proteowizard.sourceforge.io/download.html), although that is not very fast and indeed results in huge files. Side note, combining AlphaTims with msconvert serves as a "fast" work around to identify ddaPASEF data with sage:

**Install environment**
```bash
conda create -n test python=3.10 -y  # Creates a new cond anv
conda activate test # Activates new conda env
pip install alphatims # Installs AlphaTims
conda install -c bioconda proteowizard -y # Installs Proteowizard, which contains the msconvert command
conda install -c bioconda -c conda-forge sage-proteomics -y # Installs Sage
```
**Running analysis**
```bash
conda activate test # Activates new conda env
alphatims export mgf sample.d # Converts a sample.d folder with raw DDA data to a sample.mgf in the same folder as sample.d 
msconvert sample.mgf # Converts a sample.mgf file into a sample.MzML file in the same folder as the active working directory
sage config.json # Runs sage with parameters from config.json
```

lazear, 2023-08-31:
Hi Sander, 
Thanks for the response - that makes sense (I'm still clearly wrapping my head around IM stuff 😃). 
I appreciate the customized example haha! I will give it a shot with the workaround.