MannLabs/alphatims #277: [Question] Precursor masses from XML settings are not present as indices in raw data

baderd, 2023-09-13:
Dear alphatims developers,

I cannot open the file “sample_414.d” in the alphatimsGUI it gives the error “ValueError: slice step cannot be zero”.
Log file attached:
[log_20230911152500.txt](https://github.com/MannLabs/alphatims/files/12594824/log_20230911152500.txt)

When I open the file in python, it works.
```
timstof_data = alphatims.bruker.TimsTOF(tmp_file.as_posix())
tab_raw = timstof_data[:,:,:,:]
```

However, looking at the `precursor_indices` column leaves me puzzled. It contains only the value “0”, but I specified 4 values in my settings.

Number of unique values and default print:

[image]

First 10 rows of all columns:

[image]

Furthermore the “timstof_data.precursors” data frame is None. The MS1 filter values are also shown in the Bruker analyzer software. Screenshot from “sample_414.d\414.m\microTOFQImpacTemAcquisition.method” (and I measured in positive mode). 

[image]

**Question**

Do I look in the wrong column? Are the MS1 filter masses somehow encoded in another index column?

I am looking forward to your suggestions :-)

sander-willems-bruker, 2023-09-13:
Hi @baderd , welcome to the AlphaTims community;). 
* With respect to the first question about the gui; This bug should be fixed in the latest development branch, although I believe it is unrelated to your actual issue .
* Your second question is very difficult for me to assess without having the raw data at hand. My best guess at this moment is that you do not have dia/dda-PASEF data or use some settings which I have not seen before. Would you be able to share it with me so I can investigate?

baderd, 2023-09-13:
Hello @sander-willems-bruker ,

Thank you very much for the fast reply! 
Indeed, the GUI is no actual problem for me. Good to know this will be fixed as well :-) 
We will discuss internally, how we can give you access.

swillems, 2023-09-26:
Dear @baderd , thank you for sharing the data with me. It turns out that this unfortunately is an acquistion method that is not supported in AlphaTims. AlphaTims was originally developed for proteomic samples acquired in diaPASEF or ddaPASEF on a TimsTof. While the former is only an implicit constraint (I have seen several non-proteomic samples being processed with AlphaTims), it does generally imply an LC-ESI setup. The latter however is a hard constraint, if is not ddaPASEF or diaPASEF the data simply cannot be read properly by AlphaTims. Prm support might be possible in the future if time permits, but at the moment there are no concrete plans. I would be more than happy to accept a PR that actually implements this for AlphaTims though!

baderd, 2023-11-16:
With the awesome help from @sander-willems-bruker 👑 , I found a workaround for me.

Since we do not have the standard setup, we are lost at this [if-else step](https://github.com/MannLabs/alphatims/blob/master/alphatims/bruker.py#L186-L215). Our data has the value "2" in the MsMsType column of the Frames table, which is not considered. 

However, a table with the info we need is present: **FrameMsMsInfo**. It contains the "Frame" index and the "TriggerMass" which are our MS1 precursor filter values. 

You can nicely query your `analysis.tdf` files with the SQLite browser. In case you search more columns.

HTH,

Daniel

sander-willems-bruker, 2023-11-16:
Gald that you managed to find somethinbg that works! If the only thing we need to do is checking for the value `2` in the `MsMsType` column, I might be able to make a quick PR to simplify things and skip the workaround alltogether! To (hopefully soon) be continued...

closed 2025-03-06 by mschwoer