MannLabs/alphatims #278: Indexing is broken: Do you only accept sorted integer lists?

picciama, 2024-01-16:
**Describe the bug**
Indexing does not work as expected with an iterable. Wrong elements are returned and correct elements are missing.
I think you do not actually support iterables but only sorted iterables.

**What I want to do**
Given a combination of frames and precursors, I want to only get the fragments matching to the provided pairs (frame, precursor)

**To Reproduce**
There are plenty of ways to break it but this is what's working consistently for me:

```python
data = TimsTOF(<path_to_.d>)
matching_df = data[[2,2,3], :, [1,3,1]]
matching_df
```
I.e., create an object and select frame 2 and 3 where precursors 1 and 3 should be returned for frame 2 and precursor 1 for frame 3. 

**Expected behavior**

A dataframe should be returned where the frame_indices and precursor_indices in individual rows should be matching pairs like they were indexed. In this case:

frame_indices, precursor_indices
2, 1
2, 3
3, 1

**Actual behaviour**
Every precursor except 1 is ignored. In case of more indices, it also happens that random precursors are selected that aren't even in the list of indices.

**Screenshots**
[image]

Here you can see how the pair (2,3) is missing in the dataframe.

**Version (please complete the following information):**
 - latest stable alphatims within conda environment, installed with pip
 - If no log is available, provide the following:
   - Platform information
     - system    Ubuntu
     - release  22.04.1
     - machine   x86_64
     - processor Intel(R) Xeon(R) Gold 6148
     - cpu count 160
   - Python information:
     - alphatims version 1.0.8

**Additional context**
Add any other context about the problem here. Attached log files or upload data files if possible.

sander-willems-bruker, 2024-01-16:
Dear @picciama , you are fully right that indeed these indices need to be sorted (you can see this is in the docs if you do `help(data)`, specifically the lines: https://github.com/MannLabs/alphatims/blob/master/alphatims/bruker.py#L627-L641).

If your interest is in retrieving fragments coming from the same precursor, you can check out the "Advanced DIA example" in the Jupyter tutorial notebook (https://github.com/MannLabs/alphatims/blob/master/nbs/tutorial.ipynb) to get some inspiration (i.e., the `inspect_peptide` function). This shows how you can retrieve all fragment indices that fall within certain tolerances, although I'l admit this is not necesarilly the most efficient way to do it.

As a second alternative, check out the `filter_indices` function that is used under the hood when you call `__getitem__` (https://github.com/MannLabs/alphatims/blob/master/alphatims/bruker.py#L2961). This might fit your needs if you care about performance, but it is much more low level in terms of input parameters.

picciama, 2024-01-17:
Hi @sander-willems-bruker , I already had the suspicion that this is desired behaviour but somehow missed the IMPORTANT despite it being in all caps... I changed the title of this issue accordingly to make it a question rather than a bug. 
I looked at the tutorial and also the `filter_indices` function. The `filter_indices` function also expects an array where rows correspond to individual slices where the end index of the previous slices must be <= to the starting index of the next slice. 
In case of a precursor being present in multiple frames, this is difficult to achieve because I cannot combine a precursor and frame selection. I would have to sort both and remove duplicate indices and hence retrieve way more data than necessary as it provides me with all combinations of frame and precursor indices. That leads to high memory requirements.

I will go with the following workaround for now, similar to how things are done in the  `inspect_peptide` function in the tutorial.

```
list_of_dfs = []
for frame in frames:
    precursors = sorted(precursors_for_frame[frame])
    list_of_dfs.append(data[frame, :, precursors])
df = pd.concat(list_of_dfs)
```

The problem is that this gets even more complicated when filtering for a scan_idx range in the next step.
I will check if is better to simply retrieve the entire dataframe and merge with pandas on the required frames, scans, and precursors or looping through and concatenating.

Do you think it would be possible to create a feature like fancy indexing with unsorted indices in the future?
It would also help, if one could even just read specific frames, scans and precursor combinations from the start. I have found some low level functions like https://github.com/MannLabs/alphatims/blob/96d10a8927dbeeeef0ab1a835bd742b948eec8f7/alphatims/bruker.py#L1059 that could technically be customized to read only required information as it uses https://github.com/MannLabs/alphatims/blob/96d10a8927dbeeeef0ab1a835bd742b948eec8f7/alphatims/bruker.py#L133 to provide the indices from the metadata.

I suggest to separate the initialisation of the AlphaTims object from the actual reading process, i.e. offer the possibility to create an AlphaTims object first, then use its reading functions manually like:

```
data = AlphaTims()
data.read(frame_ids, scan_ids, precursor_ids, ...)
```
For now, this can be realised by creating a new class inheriting from AlphaTims and writing such a read function that calls the internal functions of the original AlphaTims class. I can try this and share some code with you in case you think that might be a good feature.

sander-willems-bruker, 2024-01-17:
Dear @picciama , happy to hear there is at least some workaround that helps you to tackle your issue. With respect to the fancy indexing, there is no direct plan to implement this.

AlphaTims first and foremost is a way to access data, whereas your use case also deals with organizing the data to a large extent, basically you are trying to do two things at once. TimsTOF data in essence is always "sorted" by design, as we deal with TOF data and he natural order is thus when it was detected (with rt> im > mz as a natural preordering). Note that the `filter_indices` in itself is very efficient and allows full parallelization as well. Moreover, this function return a (sorted) list of indices for raw ions (which is not so memory intensive). Getting "the right" data like this would be the first step of what you try to achieve. The second step for you would then be to organize that data and understanding it. Converting raw indices to a dataframe also happens as a second step in the `__get_item__` function through the underlying `as_dataframe` function (https://github.com/MannLabs/alphatims/blob/master/alphatims/bruker.py#L1741). Note that this function actually DOES have a `raw_indices_sorted` flag that properly deals with differently organized data. In general this is my preferred approach to deal with the data: get what I need first as efficienty as possible, organize it later.

You final suggestion to postpone data reading until needed is an interesting one. For AlphaTims this is not directly plans, but it does happen like that in e.g. TimsRust (https://github.com/MannLabs/timsrust). As a work around to not have all data in RAM, you can always convert your raw data into an hdf file with e.g. a CLI call `alphatims export hdf sample.d`. Once this is done, Alphatims is able to load that hdf through mmapping, meaning it only fetches data drom disk when accessed (with paging done of course, I did not modify anything for the mmapping itself).

picciama, 2024-01-17:
Yes indeed, hdf conversion is exactly what I needed! 
Though it still leads to a high memory footprint during conversion of course and mmapping only working with uncompressed hdf (with compression it fails here: https://github.com/MannLabs/alphatims/blob/96d10a8927dbeeeef0ab1a835bd742b948eec8f7/alphatims/bruker.py#L1281 with the error `OSError: Array /raw/_intensity_values cannot be mmapped. Perhaps it is compressed or chunked?` ).

I can live with that though, since our users are  likely having the large .d / hdf files on their NAS and can convert it once for later usage in multiple runs. This is actually pretty neat as accession time is very low.

According to yxour proposed strategy of first getting indicies, then organising data (also for other people who might need a general solution):

1. First create the TimsTof object without slicing as a dataframe from an uncompressed hdf:
```
data = alphatims.bruker.TimsTOF("myAwesomeTimsExperiment.hdf", slice_as_dataframe=False)
```
2.  create a list of desired (frame id, precursor id) pairs, and sort first by frame id, then by precursor id.
```
df_pasef = df_pasef.sort_values(["FRAME", "PRECURSOR"])
df_pasef[["FRAME", "PRECURSOR"]][:10]
```
[image]

3. group together all such pairs that form a closed group that can be written in ascedning order, i.e. all combinations of frame and precursor in this group need to be retrieved.
```
df_frame_group = df_pasef[["FRAME", "PRECURSOR"]].groupby("FRAME", as_index=False).agg({
    "PRECURSOR": tuple,
}).groupby("PRECURSOR", as_index=False).agg({
    "FRAME": tuple
})
df_frame_group[["FRAME", "PRECURSOR"]][:3]
```
[image]

4. loop through all groups, retrieve the list of all raw indices and convert to a dataframe only once at the end with the flag `raw_indices_sorted=False`. Retrieving raw indices in groups like I do here, may already guarantee that they are sorted, but I haven't proven this. This is still pretty fast though.

```
raw_idx = []
for frames, precursors in zip(df_frame_group["FRAME"], df_frame_group["PRECURSOR"]):
    raw_idx.extend(data[frames, :, precursors])

df2 = data.as_dataframe(
    raw_idx, 
    raw_indices=False,
    frame_indices=True,
    scan_indices=True,
    quad_indices=False,
    tof_indices=False,
    precursor_indices=True,
    rt_values=True,
    rt_values_min=False,
    mobility_values=True,
    quad_mz_values=False,
    push_indices=False,
    mz_values=True,
    intensity_values=True,
    corrected_intensity_values=False,
    raw_indices_sorted=False
)
```

This way, one can get exactly what's needed and the memory footprint went from around 12 GB down to around 3-4 GB, simply by only really loading whats necessary.

sander-willems-bruker, 2024-01-17:
Hapy to help! The footprint indeed might remain to show large-ish, but note that the mmapping means it will "never" crash as it should remove pages from RAM if there is no space left... Also note that you do not necesarilly need to create a TimsTOF object with the `slice_asdataframe=False` flag (although in your case this might be exactly what you want!), but can just as well do a `data[...,"raw"]` to get indices only.

I'll close the issue for now as it seems like we solved it:).

All the best, Sander

closed 2024-01-17 by sander-willems-bruker